Tom-evnut/TeslaBMSV2 #3: only cosmetical -> double display value

stxShadow, 2019-07-11:
Serial2.print("firm.val=");
  Serial2.print(firmver);
  Serial2.write(0xff);  // We always have to send this three lines after each command sent to the nextion display.
  Serial2.write(0xff);
  Serial2.write(0xff);
  Serial2.print("firm.val=");
  Serial2.print(firmver);
  Serial2.write(0xff);  // We always have to send this three lines after each command sent to the nextion display.
  Serial2.write(0xff);

Double serial output of the firmware version to Nextion Display.

It would be quite nice to have a page with the actual cell voltages .... 

Maybe i can contribute that to your project ? 

Jens

stxShadow, 2019-07-11:
just to simplify the write(0xff):

you can do:
byte NextionTerminator[3] = {255,255,255}; // global declaration

Serial2.print("firm.val=");
Serial2.print(firm);
Serial2.write(NextionTerminator,3);

etc ..... should shorten the whole thing

tomdebree, 2019-10-29:
Fix implemented, will think about the shortening of coding with the array.

closed 2019-10-29 by tomdebree