Tom-evnut/TeslaBMSV2 #4: SMA Sunny Island CAN support?

chron0, 2019-08-20:
Thanks for making and sharing this great open-source BMS. I'd love to build a second life EV bat pack with a Sunny Island as basic off-grid solution. So far I could only find batrium and rec bms with support for the SI. In the .ino it says device "Victron/SMA" but I'm not sure whether that means that the SMA Sunny Island is supported or not. There are a lot of them out there, I'm sure there is more "demand" for a FOSS BMS solution working with them.

tomdebree, 2019-10-29:
The canbus works with the 48V systems provided by SMA, so yes alos Sunny Island

closed 2019-10-29 by tomdebree

arvpaul, 2019-10-29:
[image]
My system is working fine with the SimpBMS, Sunny Island 6.0H, and 30KW/h Tesla modules from Mercedes B Electric :-)

chron0, 2019-10-29:
Thanks for the update and the real world config image :)

dalklein, 2020-12-17:
> [image]
> My system is working fine with the SimpBMS, Sunny Island 6.0H, and 30KW/h Tesla modules from Mercedes B Electric :-)

Nice!  Congrats on the nice looking setup!    I see you have a Sunny Remote connected to the Sunny Island.  Any chance that you have a dump or log of the canbus messages while using the Remote to change menu settings in the Island?  It would be very useful to be able to program setting changes to the inverter based on a schedule, from a raspberry pi or other device, with a canbus adapter to the Sunny Islands.      Thanks!!

arvpaul, 2020-12-20:
Hi ! 

I dont have any log of the canbus messages.. 
I will let you know... if I manage to log the canbus som time...

Arve