Tom-evnut/TeslaBMSV2 #5: Bug "R" -> restart not workgin

stxShadow, 2019-09-04:
Firmware Version : 190729

Pressing "R" to restart the BMS is not working. The serial connection is reseted but the device is not coming back to life.

stxShadow, 2019-09-05:
Follow up -> to get reboot working change 
#define CPU_REBOOT (_reboot_Teensyduino_());
to 
#define RESTART_ADDR       0xE000ED0C
#define READ_RESTART()     (*(volatile uint32_t *)RESTART_ADDR)
#define WRITE_RESTART(val) ((*(volatile uint32_t *)RESTART_ADDR) = (val))
#define CPU_REBOOT (WRITE_RESTART(0x5FA0004));
-> restart is working fine after that

BobbyBleacher, 2019-09-23:
Is this right? It won't compile for me.

stxShadow, 2019-09-24:
works quite nice for me. What error are you seeing ? My own little fork looks like this -> line 33 to 42:

#include <SPI.h>
#include <Filters.h>//https://github.com/JonHub/Filters

#define RESTART_ADDR       0xE000ED0C
#define READ_RESTART()     (*(volatile uint32_t *)RESTART_ADDR)
#define WRITE_RESTART(val) ((*(volatile uint32_t *)RESTART_ADDR) = (val))
#define CPU_REBOOT (WRITE_RESTART(0x5FA0004));

BMSModuleManager bms;
SerialConsole console;

BobbyBleacher, 2019-09-25:
TeslaBMSV2: In function 'void menu()':
TeslaBMSV2:35: error: lvalue required as left operand of assignment
 #define WRITE_RESTART(val) (((volatile uint32_t *)RESTART_ADDR) = (val))

                                                                 ^

C:\Users\Bobby Martin\Dropbox\arduino-1.8.8\portable\sketchbook\TeslaBMSV2\TeslaBMSV2.ino:36:21: note: in expansion of macro 'WRITE_RESTART'

 #define CPU_REBOOT (WRITE_RESTART(0x5FA0004));

                     ^

C:\Users\Bobby Martin\Dropbox\arduino-1.8.8\portable\sketchbook\TeslaBMSV2\TeslaBMSV2.ino:2252:9: note: in expansion of macro 'CPU_REBOOT'

         CPU_REBOOT ;

         ^

tomdebree, 2020-04-02:
Fix implemented, works now.

closed 2020-04-02 by tomdebree