Tom-evnut/TeslaBMSV2 #8: Nextion stat.txt stuck to "boot" in ESS Mode

stxShadow, 2019-09-04:
The stat.txt field in nextion displays never geht to "ready" in ESS Mode. 

Value is stuck in "boot".

stxShadow, 2019-09-06:
As far as i see there is no other output than "boot" if in ESS Mode.
Maybe you should send something like "ESS Aktive".

stxShadow, 2019-09-24:
changed the code like this:

void dashupdate()
{
  Serial2.write("stat.txt=");
  Serial2.write(0x22);
 
  if ((settings.ESSmode == 1) && (bmsstatus != Error))
  {   
      //In ESS mode print only ESS Active / Error messages to the nextion display
      Serial2.print("ESS Active");
  }
  else
  {
    switch (bmsstatus)

works for my case now

tomdebree, 2019-10-29:
Sorry the screen was mainly meant for Vehicle operation so none ESS mode, will roll this change into code.

closed 2019-10-29 by tomdebree