Tom-evnut/TeslaBMSV2 #9: serial interface incorrect using screen or minicom

betz, 2019-09-13:
Hey,

We found out that connecting to the serial console on debian using screen or minicom has issues.
Connecting on the same laptop thru the arduino IDE serial monitor works fine.
The issues are that we were unable to change some settings.

A guess is that it has something to do with windows/unix newline, which arduino IDE probably handles better?

Cheers,
Tom

stxShadow, 2019-09-13:
yes. has something to do with the serial port mode. minicom / screen use character wise communication while simp bms seems to use line wise input. I found no way to change settings except this way: 

stty -b 115200 -F /dev/ttyACM0
stty -F /dev/ttyACM0 -cooked
cat /dev/ttyACM0 
in another screen: 
echo "3 70" > /dev/ttyACM0 (for instance)

Explanation to the tty raw and cooked mode:
https://unix.stackexchange.com/questions/21752/what-s-the-difference-between-a-raw-and-a-cooked-device-driver

Hope it helps.

Jens

betz, 2019-09-14:
looking at the code, guess that's where to start looking
https://github.com/tomdebree/TeslaBMSV2/blob/master/SerialConsole.cpp#L106

relevante, 2019-12-06:
I've found that on OSX with screen, typing a command sequence (e.g. 3100, if you wanted to set parameter 3 to the value 100) into a different window and then copying it to the clipboard and then pasting it in seems to let the BMS consume it all at once and work properly. For now it seems to be a decent workaround at least on my particular setup.