Tom-evnut/TeslaBMSV2 #10: Balancing abormality

BobbyBleacher, 2019-09-20:
See the screenshot there, Cells 20 and 21 keep bouncing back and forth between unusually high and low values. Unsure of the cause. Not documentation found to support this behavior.
[image]

BobbyBleacher, 2019-09-21:
If anyone else sees this, I found out way it probably is.

It's what happens when there is a bad connection to a cell, it senses a common mode voltage of two cells split between the two readings... ((4.87 + 2.57)/2) = 3.72

tomdebree, 2019-10-29:
Alot of issues happen due to faulty wiring it is a pain to track down.

closed 2019-10-29 by tomdebree