Tom-evnut/TeslaBMSV2 #15: PreCharge stays on

Webwanze, 2020-02-06:
Hello,

in my opinion PreCharge should be switched off after main contactors are activated. 

- saves energy and wear
- prevents possible damages if main contactor fails 

Thomas

tomdebree, 2020-02-06:
It goes off. Check your configuration.

closed 2020-02-06 by tomdebree

Webwanze, 2020-02-06:
I‘m in ESS mode. 
J4/2-GND
J4/13-PreCharge Relay

It stays on. Cannot find any settings. (Latest version)

tomdebree, 2020-02-06:
Why are you running ESS mode and using pre charge? Read the manual…..

[cid:image003.png@01D5DD23.E0385640]

Webwanze, 2020-02-06:
Its a big battery for my home. This points to ESS. 
Also my SMA SunnyIsland should be precharged...

When using default mode:
After key in signal both contactors are closed the same time and precharge switches to off after a few seconds. 
I would expect first precharge - then main contactor - then precharge off.

tomdebree, 2020-02-06:
Sure but are you using a SimpBMS?

Read the manual, ESS mode is NOT using precharge. An ESS installation requires manual precharging.

It does do this in the Normal mode. It does not require key in signal for ESS mode.