Tom-evnut/TeslaBMSV2 #17: Temperatur t0 prefix not shown on display

stxShadow, 2020-02-25:
Hi Tom,
t0.txt -> +- not shown

you forgot 

    Serial2.write(0x22);
    Serial2.write(0xff);
    Serial2.write(0xff);
    Serial2.write(0xff);

after

Serial2.write(0x22);
    if (bms.getAvgTemperature() > 0)
    {
      Serial2.print("+");
    }
    else
    {
      Serial2.print("-");
    }

tomdebree, 2020-03-03:
Turns out there was no need for this.

closed 2020-03-03 by tomdebree

stxShadow, 2020-03-03:
agree .... temperature was not necessary 
but you also removes t7 (plus / minus) current ..... i find this very useful in ESS environments.

tomdebree, 2020-03-03:
Turned out the display was already capable of this. the values are standard signed integers.
Please also check updated display firmware.