Tom-evnut/TeslaBMSV2 #26: SIMP BMS damaged - looks like overheating.

jono230981, 2022-07-15:
I have been using the SIMP BMS with Victron CX for a number of months with the SIMP connect to the 'VE - CAN' port and a Victron terminator in the second port on the CX.  Running at 500kbps with the CX set to CAN-BUS BMS on the VE CAN port.  Its been working with no real issues. 

 I purchased and additional Victron Solar charger with VE CAN connectivity , after a bit of trial and error I set the VE-CAN port on the CX to VE-CAN & CANBUS 250kbps.  I also set the speed on the Simp to match at 250kbps.  This seemed to work OK and I could see both the Solar charger and SIMP on the Victron CX.    An hour later the SIMP BMS goes offline and investigating it seem to have burnt out 'C13'  please see attached picture.  As this had been running fine for a number of month prior to the change I can only assume I have run into hardware/configuration compatibility.  Hence I'm
[image]
[image]
 reaching out to you for some assistance. 

Thanks

jono230981, 2022-07-15:
Hi,

I'm going to assume the 12v power supply is fine, as its been ok for a few months and its too much of a coincidence that i connect the solar charger and it experiences the issue an hour later. 

Victron Smartsolar 250/85 VE-CAN

jono230981, 2022-07-15:
[image]

jono230981, 2022-07-15:
I was hoping to just replace the capacitor , I've removed the burnt remains of the original but unfortunately it looks like it's damaged the solder pad on the board, so I'm guessing it's not repairable. [image]

Tom-evnut, 2022-07-15:
Please check the BMS still boots with the VEcan cable disconnected.

jono230981, 2022-07-15:
Im powering it from a 12v/240 adapter.  By loads do you mean DC contactors etc?  If so No loads on the BMS. ie its just pulling the telemetry from the slaves and (was) passing it via VE-Can to the Victron CX. I do have the Nextion screen but I assume that only draws a few 100 mA.

Tom-evnut, 2022-07-15:
I would suspect that there ahs been a power spike of some sorts that got through the 12V/240V adapter.

How hot is the area where the BMS is mounted?

jono230981, 2022-07-15:
Its not hot at all.   What does that capacitor do?

I have just reconnected the RJ45 back to the Victron CX. ( I've left the new solar charger disconnected ) and the CX is receive the SIMPBMS data .

Tom-evnut, 2022-07-15:
It smooths the 12V input and buffers it. 

My theory right now is the AC/DC adapter used is being used at too low a power and thus has alot of ripple or there was a voltage surge or something and it passed it onto the caps.

jono230981, 2022-07-15:
Is it safe to use with the cap missing?  Is it possible to rig up something off the board to do the same function?

Tom-evnut, 2022-07-15:
Its is fine to use like this. I would evaluate the 12V provided by the AC/DC adapter with a scope to see how dirty it is and possibly look at going for a better one or another way of getting 12V.

jono230981, 2022-07-15:
Ok brilliant. I will look into that.  Thank you for your help and quick responses.