Tom-evnut/TeslaBMSV2 #27: Missing tesla slave bms reads

Jcr787, 2022-10-14:
Hi, Im running an variation of your code and sometimes I get missing module reads which disconnects main contactor due to Series cell fault. Some other times, at power up, I get a series Cell fault. When I check in the debug window, it says a cell number lower than the saved pack configuration (missing reads). It comes back online after  "cells present reset".

Have you seen something similar in the past?

Tom-evnut, 2022-10-14:
As you have made a variants I cannot directly know what you changed. Series cell fault is due to missing cells reported. On a tesla system this is hard to achieve I believe unless your communication for some of the modules gets corrupted. You need to log and look what happens to the received information if there are any drops. Cells present reset means it resets the amount deemed present, this gets set a few seconds after booting to allow the system to stabilize.

Jcr787, 2022-10-14:
Im running the code in an Arduino due and I changed only the necessary. For the most par it runs fine but Im getting random drops due to missing reads. It shows at the pack with 0 volts across the cells.

Tom-evnut, 2022-10-14:
So corrupted packets being thrown out causing zero reads, you can remove the clearing/zeroing of the historical data function  bms.clearmodules();